uwrobotics/MarsRoverFirmware #2: Create non-blocking NeoPixel driver

jetkov, 2019-09-26:


jetkov, 2019-10-04:
https://learn.adafruit.com/dma-driven-neopixels

Aethelred-The-Ready, 2019-10-04:
I found a way to do this super easy, but I would need to add a little bit of external processing equipment, or I could do it in a slightly more complex way that would use 4x the RAM. Either way, the code is actually quite simple, so it should be all right.
The processing equipment is basically described here, a little pulse generator https://github.com/benheck/SPI-Neopixel/blob/master/schematic_image.png. Which should I go for?

jetkov, 2019-10-04:
That's a pretty neat solution! I would just go with the 4x method though (are you talking about SPI bit banging?) since we have a ton of RAM with the F446RE and we can test it out easily with the hardware we already have.

Aethelred-The-Ready, 2019-10-04:
Yup, bit banging, the tolerances for neopixels are low enough. We will be fine for RAM (we could run more matrices than we could transmit to with 128K). I am a little worried about the calculations involved, but i'll check them later today with some neopixels. It has to be rerun every 6 ms.

jetkov, 2019-10-04:
Great. Also don’t feel rushed on this, we have plenty of time.

Aethelred-The-Ready, 2019-11-29:
Tried just normal SPI in mbed, too slow by about half, would consistently just revert to inactive high every 4 bits which would confuse the LEDs.
Started to build out a pulse generator to take in a slower SPI signal and convert it into the proper LED signal, but it would require around 3~4 ICs on a PCB, which is just a hassle.
Two ideas:
Get down into the lower level code, speed up the DMA to work at 2.9 MHz, at least for these signals
or
Use a small arduino to take in an SPI or UART or some other signal, and then output the pulses properly

wmmc88, 2019-11-30:
Do you know what the limiting factor is in the spi bit banging method?

Aethelred-The-Ready, 2019-12-02:
No clue, just that the SPI DMA offered through mbed is too slow

wmmc88, 2020-07-09:
@jetkov closeable?

jetkov, 2020-07-09:
> @jetkov closeable?
yes

jetkov, 2020-07-09:
(a neopixel implementation however neopixels will no longer be used on the rover)

closed 2020-07-09 by jetkov