uwrobotics/MarsRoverFirmware #3: Create NeoPixel matrix display control library

jetkov, 2019-09-26:
We are planning to use a 8x8in 2D array of NeoPixels as our LED indicator for the automation task. A library will be needed with function calls to setup the matrix and display red, blue, and green on this display at the minimum.

Nice to have:
- Ability to flash on and off with set colour
- Ability to display 1 or 2 alphanumeric characters

laineee, 2020-05-16:
Can I work on this?

wmmc88, 2020-05-16:
@jetkov correct me if im wrong but this issue should be closed cuz we're moving to the led matrix board and its not addressable.

jetkov, 2020-05-16:
Yes this issue should be closed, but I'll tag you in the issue for the LED board @laineee. It should be a very simple first task.

closed 2020-05-16 by jetkov