uwrobotics/MarsRoverFirmware #5: Create absolute PWM encoder driver with support for rollover

jetkov, 2019-09-28:


jetkov, 2019-10-02:
In the 2019 season a PwmIn library from the MBED community was used. It was modified to enable a configurable rolling average on the input, since our signal was relatively noisy. The signal capture and processing could be improved. 

The new driver should account for when the encoder rolls over 360 degrees (its full range) so that the driver outputs the true degrees of rotation (ie. -20 or +390 degrees). Currently it just rolls over to 0 when 360 is exceeded. This may not be necessary if we do not care about when an encoder rolls over.

tandronescu, 2019-11-10:
No longer need to do this since we have the limit switches on the arm to serve as the initial calibration point.

closed 2019-11-12 by jetkov