uwrobotics/MarsRoverFirmware #6: Create better PID controller library

jetkov, 2019-09-28:


tandronescu, 2019-11-15:
Main improvement will be adding an Autotuning library which generates tuning parameters using a relay method of oscillating the output. By analyzing how far apart the peaks are, and how big they are in relation to the output changes, the Autotuner can tell the difference between one type of process and another. See explanation: http://brettbeauregard.com/blog/2012/01/arduino-pid-autotune-library/ 

Other suggested improvements looking at the code:
-inside the setInputLimits function there will be a division by 0 in the first run thru since inSpan_ doesn't get set until the end of the function. We should probably set this at the beginning
-the variable realOutput_ is set to 0 and never updated. It is used in the reset() function, but I'm not sure what the intended purpose of this is
-when we reset(), we set accumulated error to 0. This could cause significant output spikes if switching from manual mode to auto mode while the driver is running and has a high output. Instead, it may be more practical to set the accumulated error to the current output.
-Although this doesn't cause functional issues, for changing tuning parameters on the fly it may be cleaner to pull the Ki term into the accumulated error instead of re-scaling the accumulated error. This is debatable, but it would be less code to just do this change

jetkov, 2020-01-28:
Fixed in PR #29

closed 2020-01-28 by jetkov