uwrobotics/MarsRoverFirmware #18: Investigate using timer hardware to capture PWM information

jetkov, 2019-10-25:
Measuring a PWM signal period and duty cycle using timers in capture mode should be more precise and efficient than using rising/falling edge interrupts.

FK3wong, 2020-05-21:
@jetkov I'm looking on taking up another task as I've put a short pause on the current sensor task I was working on. Can you give me some more information about this issue? I'm a bit unsure about how this is used on the rover and would help give me context if I'm able to take it.

jetkov, 2020-05-22:
Hey @FK3wong, the purpose of this and issue #19 is to use the STM32 timer functions for measuring duty cycle, and quadrature decoding, instead of using rising/falling edge interrupts and the system timer. This should free up system resources a bit and allow more asynchronous operation: https://www.st.com/content/ccc/resource/training/technical/product_training/c4/1b/56/83/3a/a1/47/64/STM32L4_WDG_TIMERS_GPTIM.pdf/files/STM32L4_WDG_TIMERS_GPTIM.pdf/jcr:content/translations/en.STM32L4_WDG_TIMERS_GPTIM.pdf

FK3wong, 2020-05-22:
Possible useful resources:
https://www.st.com/resource/en/application_note/dm00236305-generalpurpose-timer-cookbook-for-stm32-microcontrollers-stmicroelectronics.pdf 
https://www.st.com/content/ccc/resource/training/technical/product_training/c4/1b/56/83/3a/a1/47/64/STM32L4_WDG_TIMERS_GPTIM.pdf/files/STM32L4_WDG_TIMERS_GPTIM.pdf/jcr:content/translations/en.STM32L4_WDG_TIMERS_GPTIM.pdf 
https://www.st.com/content/ccc/resource/technical/document/user_manual/2f/71/ba/b8/75/54/47/cf/DM00105879.pdf/files/DM00105879.pdf/jcr:content/translations/en.DM00105879.pdf page 868