uwrobotics/MarsRoverFirmware #23: Develop better CAN driver

jetkov, 2019-11-12:
- We should implement handler functions that make it easier to send and receive messages with a payload of a single variable datatype
- Should handle two CAN interfaces with a failover mode

jetkov, 2019-11-18:
https://drive.google.com/file/d/1MSh121hMTI4BJESQDF_F4vjrMP9cAI3t/view?usp=drivesdk

wmmc88, 2019-11-30:
Just some thoughts:
- default ifnames will be can0 and can1
- How often would we realistically miss either heartbeat?
- when u write `awk`, you mean `ack` right?

wmmc88, 2019-11-30:
looks solid. do we have people to implement the jetson and firmware sides of this?

jetkov, 2019-12-01:
> Just some thoughts:
> 
> * default ifnames will be can0 and can1
> * How often would we realistically miss either heartbeat?
> * when u write `awk`, you mean `ack` right?

* I wrote CAN1 and CAN2 because that is what STM named them in the HAL. So on the firmware side at least we will call them this so not to cause confusion within the FW levels. But Jetson side can do whatever
* We shouldn't miss a beat unless our T is very small or we have a bug or an electrical disconnect
* `ack` yes. Was using `awk` too much on the cli at the time lool

wmmc88, 2019-12-01:
we can actually just name the interfaces on the jetson to `can1` and `can2`

cindyli-13, 2021-06-04:
Closing this as we're going by a new CAN architecture

closed 2021-06-04 by cindyli-13