uwrobotics/MarsRoverFirmware #44: Add PID configuration API to Actuator class

jetkov, 2020-01-31:


yreda1999, 2020-04-06:
this looks like fun. is it still open? if so could u explain a little more abt it

jetkov, 2020-04-14:
Hey Younes, sorry for the late reply. We want to be able to tune the PID parameters for each actuator over CAN, so we need to add an API to the actuator class to be able to do this... not just to update the parameters but also handle applying the parameters safely to not create a situation where the joint can be damaged.

jetkov, 2020-04-14:
If you take a look into the actuator class you can see that it uses the PID class to control the joint in the PID modes, but there is not exposure to update the parameters on-the-fly at the moment.

yreda1999, 2020-04-17:
are we looking to change the tuning parameters on the fly? I am referring to the individual params of proportional, integral, differential.

wmmc88, 2020-04-17:
Yeah. End goal is a rqt_reconfigure node that sends the commands to change to new p, I, d values.

yreda1999, 2020-04-17:
ah i see, so we wanna be doing the autotune calculations on the jetson instead of on the pcbs. is that right?

yreda1999, 2020-04-24:
@wmmc88 are we going to be using a mix of auto tuning along with manual tuning (on a per actuator basis). Will an actuator have to transition between manual and automatic tuning once a msg asking to update a pid parameter is received? I want to make sure i understand our intended use case so i dont miss any edge cases.

yreda1999, 2020-04-24:
or rather are we not looking to run actuators in manual tuning mode but rather in adaptive tuning mode where we suggest pid params on the fly: https://playground.arduino.cc/Code/PIDLibraryAdaptiveTuningsExample/

wmmc88, 2020-04-24:
Doubtful that we'll adaptively change pid gains within any given motion. But I'd defer that to @jetkov

wmmc88, 2020-04-26:
@jetkov where is the autotune code and how exactly does it work? how does it fit into this issue?

jetkov, 2020-04-27:
@wmmc88 just had a call to @yreda1999 to talk about this.... Auto-tune code was ported from another MBED project by @tandronescu and hasn't yet been tested on a limited-range actuator like the arm. For this issue just being able to tune the PID params over CAN is the main goal but secondarily to manage auto-tuning.

cindyli-13, 2021-06-04:
Closing this as we're using ODrive PID control

closed 2021-06-04 by cindyli-13