uwrobotics/MarsRoverFirmware #47: Complete Gimbtonomy controller application

jetkov, 2020-01-31:
Will be complete when:

- [ ] Can accept incoming CAN messages
- [ ] Handles CAN commands to control the camera gimbal
- [ ] Handles CAN messages to control the NeoPixels
- [ ] Provides feedback to Jetson on controller status

FelixWLWong, 2020-04-30:
@jetkov How difficult would this task be? I think this task is pretty interesting but might take me a while depending on its difficulty. Instead of sending messages to the neopixel would we instead be sending handling CAN messages for the LED board (since the neopixels are being phased out)?

jetkov, 2020-05-02:
Hey @FelixWLWong @KrillNye is already working on this, you can check it in his PR. I don't know too much about the new LED board but I don't think it uses CAN. @KrillNye can you create an issue for writing the driver for the LED and add the information for that?

jetkov, 2020-05-29:
@qinyang-bao is working on the gimbal control, @laineee is working on the LED board control.

closed 2021-06-04 by cindyli-13