uwrobotics/MarsRoverFirmware #50: Upgrade to MBED OS 5.15.0

jetkov, 2020-02-09:
Current MBED OS version is relatively outdated

wmmc88, 2020-04-28:
mbed-os-5.15.2 is latest version now. we're currently on 5.13.1. Is there a list of any modifications u made to the mbed source?

wmmc88, 2020-04-30:
This should probably be put on hold. MBED 6 is in beta and I think it might be released within a couple of weeks.

jetkov, 2020-05-02:
Okay cool. No modifications were made to MBED source.

wmmc88, 2020-05-02:
I recall mods to queues. Smth ab mutexes

jetkov, 2020-05-02:
Looked through the log, the only changes made were:
- Change `\n` to `\r\n\` in embedded strings
- Delete target specific files that were moved to the "targets" directory

Any other changes are part of my CAN 'draft' PR which I'll probably change with your suggestion to not modify the source.

jetkov, 2020-06-07:
MBED OS 6 has been released https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-6.0.0

jetkov, 2020-06-07:
Required for #103 @a-gratton

wmmc88, 2020-06-07:
I'll get on this

closed 2020-08-09 by wmmc88