uwrobotics/MarsRoverFirmware #53: Force rebuild of target-specific files when changing build target

jetkov, 2020-02-15:
Add recursive target file touch to Makefile.
Currently previous compiled target files could be used if not changed between switching targets causing build failure or unexpected behaviour.

Workaround: Run `make clean-all` before compiling new target.

wmmc88, 2020-04-30:
@jetkov do you know any way to reproduce this behaviour? is it just using the old target's `PeripheralPins.c` and `system_clock.c`?

wmmc88, 2020-04-30:
it looks like its still compiling the new periph pin and system clock files, but the binaries are different. i suspect that its not including the correct header files in the target folder

wmmc88, 2020-04-30:
Ok so im pretty sure what's happening is that the target header files changed when u switched targets, but there are no explicit dependencies from a c file on the target header files (only from other header files), meaning there's no dependency listed in a `.d` file, meaning make doesnt know to recompile those source files since it sees the file is already compiled. I'm not really sure what could fix this given the way the files are setup. I might try to fix this at the same time as #50, as that will also require changes to the makefile probably.

jetkov, 2020-05-02:
I did it in a jank way in 2019. Basically I stored the command line flags into a file, and then compared these previous flags with the new flags each time make was run. If the target flag was different, I `touch`ed the files that needed to be recompiled.

jetkov, 2020-05-02:
https://github.com/uwrobotics/MarsRover2019-firmware/blob/master/makefile
See `COMPILE_FLAGS_TO_TRIGGER_TOUCH`, `TOUCH_ON_COMPILE_FLAGS_CHANGE`

jetkov, 2020-05-02:
When I transitioned the 2020 builds to separate the PinMap and other files into specific target folders (instead of what I had in 2019, which was `#ifdef`s in the PinMap file) I thought that this would fix that issue because different header paths are specified. But yeah I guess make doesn't care and what matters is the dependency.

wmmc88, 2020-05-02:
Actually I think since we know its isolated to the target-specific header files, I can add a dependency on the target header files(including full path) for EACH make rule. The downside being it would force every file to be recompiled, when there are some that wouldn't need to be... but I dont think itd be easy to determine that dependency without manually listing out which src files depend on target headers

I'll test it out in the morning.

wmmc88, 2020-05-03:
FML this doesnt work. I think what happens is that make sees the new target header file prerequisites and sees that they havent been modified since the last build, so it doesnt do anything. 

I can't find or think of a different way to do it other than saving the last target run in a file like u did in 2019. Well... no way without switching to automake or cmake or bazel.

I'm gonna add back ur fix from 2019

jetkov, 2020-05-04:
Okay, thanks for looking into it. Tried to switch to cmake at the start of this season but was too painful. Maybe some day.

wmmc88, 2020-05-04:
Lol itd be nice to use cmake cuz ros also uses cmake so people should be at least a little familiar

jetkov, 2020-05-04:
I'll make an enhancement issue for it

wmmc88, 2020-05-04:
Closed by #81

closed 2020-05-04 by wmmc88