uwrobotics/MarsRoverFirmware #69: Add functionality to do OTA firmware updates

wmmc88, 2020-03-22:
Add ability to do updates over CanBus (may keep it somewhat abstract so it'd work over other interfaces?). this would involve customizing the bootloader and embedding some sort of board id into the flash of each board. This would allow us to iterate code faster (no need to fumble with flashing boards and stuff).

wmmc88, 2020-03-22:
Honestly this is like super overkill, but i'm gonna do it as a pet side project. probs will make just a seperate script or program to execute the update based off git hash or smth.

wmmc88, 2020-03-22:
No bootloader mods actually have to be done. 
https://www.st.com/resource/en/application_note/CD00167594-.pdf

can bootloader is already loaded:
https://www.st.com/resource/en/application_note/cd00264321-can-protocol-used-in-the-stm32-bootloader-stmicroelectronics.pdf

wmmc88, 2020-03-22:
https://www.youtube.com/watch?v=cvKC-4tCRgw

jetkov, 2020-04-27:
Yes bootloader is present, we just need to be able to pull the boot0 pin in the right direction (if not already). Should tell EE to add a jumper for this if we need it to be configurable.

wmmc88, 2020-04-27:
Of note for the can bootloader:

```
Once initialized the CAN2 configuration is: 
Baudrate 125 kbps, 11-bit identifier.
Note: CAN1 is clocked during CAN2 
bootloader execution because in CAN1 
manages the communication between 
CAN2 and SRAM
```

It only works on CAN2.

wmmc88, 2020-04-27:
@jetkov I dont think we want a jumper. The point of this issue would be to update firmware without needing to touch the boards at all(assume bad accessibility). 

I think we can have smth to jump to the bootloader from app