uwrobotics/MarsRoverFirmware #71: integrate clang-tidy

wmmc88, 2020-04-26:
plan is to integrate clang-tidy checks into our ci. Probably will use the same .clangtidy file [available in our ROS repo](https://github.com/uwrobotics/uwrt_mars_rover/blob/master/.clang-tidy). since we use `make`, I'll probably need to add a step to use bear to generate a compilation database json.

wmmc88, 2020-05-12:
This is proving to be a little harder than initially thought because of small differences in compilation between clang and gcc. Since ARMCC is commercially licensed, I'm going to try to get native LLVM Clang toolchain working.

younesr1, 2020-09-15:
@wmmc88 is this complete?

wmmc88, 2020-09-15:
Nope. It fell to the back burner lol. Should be easier to tackle now that we use cmake. I'll give it a shot when I got some free time