uwrobotics/MarsRoverFirmware #79: Define clear SOP and improve documentation

jetkov, 2020-05-02:


jetkov, 2020-05-02:
@wmmc88 can you link any SOP documentation for SW side, besides whats in the SW README?

wmmc88, 2020-05-02:
In terms of starting the system in ros?

jetkov, 2020-05-02:
Oh no I mean guidelines for contributing, specifications followed etc. Like how you guys standardize units for example.

wmmc88, 2020-05-02:
I don't think there's anything specific outside of following ROS conventions or anything in the readme. Haven't really formalized some sort of coding standard, nor do i think we need on formally at this point. As much as possible, I try to get people to follow this: http://wiki.ros.org/ROS/Patterns/Conventions

jetkov, 2020-05-04:
Okay cool, and your linter is based on Google's style guide right?

wmmc88, 2020-05-04:
The clangformat(mostly whitspace formatting) is Google's but with 120 column limit instead of 80. 

Clangtidy enforces some naming conventions, as well as a plethora of code style checks. Mostly the most popularly used ones with some small modification. 

U can see the clangformat and clangtidy files in the root of the ros repo

wmmc88, 2020-05-04:
For #70 and #71, I'll probably start with using the same files as a base cuz itd be nice to have uniformity between codebases. We can tweak it more once the pr is up

jetkov, 2020-05-04:
Sounds good, yeah I'd like to have uniformity as well. Obviously there will be some differences. Ex. the unit naming will not work for us because of integer types or float sizes

jetkov, 2020-05-04:
Not like that's covered by clang, but that an example of something I need to write up

wmmc88, 2020-05-04:
You'd be surprised at what u can configure clang tidy to do. You can even write custom checks for even more control.

closed 2020-05-29 by jetkov