uwrobotics/MarsRoverFirmware #83: Makefile doesn't handle filepaths with spaces, and builds fail on filepaths it does accept

a-gratton, 2020-05-05:
Error message using WSL trying to make in a filepath with spaces:

```
agratton@DESKTOP-O12R4VM:/mnt/c/Users/gratt/Documents/University/Design Teams/UW Mars Rover/MarsRover2020-firmware$ make APP=arm TARGET=arm
make[1]: Entering directory '/mnt/c/Users/gratt/Documents/University/Design Teams/UW Mars Rover/MarsRover2020-firmware/build'
make[1]: /mnt/c/Users/gratt/Documents/University/Design: No such file or directory
make[1]: *** No rule to make target '/mnt/c/Users/gratt/Documents/University/Design'.  Stop.
make[1]: Leaving directory '/mnt/c/Users/gratt/Documents/University/Design Teams/UW Mars Rover/MarsRover2020-firmware/build'
makefile:56: recipe for target 'all' failed
make: *** [all] Error 2

```

When using a filepath that has no spaces, build fails to compile with many repetitions of this type of error (tested using test-blinky and arm apps for target arm):
```
/usr/include/newlib/c++/6.3.1/bits/stl_algo.h:5536:3: error: template with C linkage
   template<typename _ForwardIterator>
   ^~~~~~~~
```

wmmc88, 2020-05-05:
those seem to be two different errors. can u replicate on master?

a-gratton, 2020-05-05:
The second error was on master, the first was from my branch which is rebased onto master.

wmmc88, 2020-05-05:
can u post the entire compile logs for the second issue? created a seperate ticket for it

wmmc88, 2020-05-05:
wrt to ur first issue:
were spaces in paths an issue before recent makefile changes?

a-gratton, 2020-05-05:
No they weren't, directories with spaces work on the most recent commit before the makefile changes.

a-gratton, 2020-05-05:
Have you posted the ticket for the second issue? I don't see it from my end.

wmmc88, 2020-05-05:
> Have you posted the ticket for the second issue? I don't see it from my end.

sorry i meant to tell you to create a seperate issue for it lol

wmmc88, 2020-05-05:
> No they weren't, directories with spaces work on the most recent commit before the makefile changes.

i'll look into it after work tonight

PR #86 (2020-05-08): fix for paths with spaces in makefile
closes #83
closed 2020-05-08 by a-gratton