uwrobotics/MarsRoverFirmware #84: Build fails with "template with C linkage" errors

a-gratton, 2020-05-05:
Many errors are produced, in the form of:

```
/usr/include/newlib/c++/6.3.1/type_traits:68:3: error: template with C linkage
   template<typename _Tp, _Tp __v>
```

And also in the form of:

```
/usr/include/newlib/c++/6.3.1/type_traits:132:3: error: template specialization with C linkage
   template<>
```

jetkov, 2020-05-07:
Can you check the following just to be sure
```
make --version # Should be v3.8.x or newer
arm-none-eabi-gcc --version # Should be v9.2.x or newer
```

a-gratton, 2020-05-07:
make is v 4.1, and arm-none is 15:6.3.1+svn253039-1build1 but apt-get tells me I have the latest version of it

wmmc88, 2020-05-07:
Did u follow the readme to use the latest version of gcc arm?

wmmc88, 2020-05-07:
@a-gratton I just built latest master on wsl. I believe u just didnt setup ur toolchain correctly. pls refer to the readme.

closed 2020-05-07 by wmmc88