uwrobotics/MarsRoverFirmware #89: LED Matrix Driver

jetkov, 2020-05-09:
- 3 RGB PWM

yreda1999, 2020-05-14:
It is preferred that this driver be non-blocking. The LED matrix needs to work alongside gimbal code. By non-blocking i mean that the function to alter the matrix should not hold up the program until the matrix is set.

jetkov, 2020-05-16:
Since it will be handled by the PWM peripherals, it should be east to make non-blocking. @lwbantoto can you confirm that the matrix pins are assigned to PWM-designated pins?

jetkov, 2020-05-16:
@lainee this is a good first task

lwbantoto, 2020-05-18:
@jetkov confirmed

laineee, 2020-05-21:
So we want to create a library with functions to be able to control the colour of the LEDs on the board. Using PWM for each colour R, G, B in different combinations to make up all the different colours and brightness. And we're using the HEX colour code to do that.

wmmc88, 2020-05-21:
More or less yes. 24bit hex colour is fine. It at minimum needs to meet requirements needed in the URC manual, which means it needs a flashing mode. Itd be nice if there were support for any arbitrary colour and a number of different patterns, pattern frequency(flashing faster vs slower for example) etc.

wmmc88, 2020-05-21:
@jetkov what do you think about getting people to have a draft pr with proposed public api, before implementation? Would help with getting people feedback earlier on in the development process.

jetkov, 2020-05-22:
Yeah that's a good idea. I already discussed this with @laineee a bit, we talked about having a function to set the hex colour and a function to set a flashing frequency.

But in yeah, it would be great to have people just write a header with at minimum the public API and submit that for review.

younesr1, 2020-09-15:
@laineee Do you mind updating this issue with your progress. It's a good way for ppl besides u and i to be in the loop for this

laineee, 2020-09-16:
I'm working on the library now that I have a clear header file from @younesr1 with what we need

cindyli-13, 2021-06-04:
Closed as this is already implemented

closed 2021-06-04 by cindyli-13