uwrobotics/MarsRoverFirmware #99: Custom Robot Controller Board Firmware

qinyang-bao, 2020-05-26:
This year we are creating our custom robot controller board, so firmware for it is required.

Preliminary design is done, see block diagram: https://app.diagrams.net/#G1zqAnAT2QxAxzYP02m8YeLTq6BpyFIup1

This "board" has a unique design: it is made up of one master board and two slave boards that should communicate between each other and a PC. While the controller parts are simple to interface with, it will be an interesting task to design/implement the communication protocol.

younesr1, 2020-08-14:
hey, was this task completed?

qinyang-bao, 2020-08-14:
The libs are merged, but we only have the main appp (untested) for the dev board. main app for any of the boards planned in the design is not written...since we don't have electrical design for them yet

qinyang-bao, 2020-08-14:
so I don't think this is completed

wmmc88, 2020-08-14:
It might be a good idea to have test apps to test new features pertinent to the controller boards, stuff like the unbuffered serial etc

qinyang-bao, 2020-08-15:
The main app we have under controller currently is for the dev board, I think it would suffice as a test app...

wmmc88, 2020-08-21:
> The main app we have under controller currently is for the dev board, I think it would suffice as a test app...

How does the dev board differ from the actual board? is the actual controller a collection of boards or a single board?

qinyang-bao, 2020-08-21:
The actual controller is a collection of boards (see block diagram at issue description). The dev board is only one board, it contains a few of each control elements (buttons, joysticks, etc.).