uwrobotics/MarsRoverFirmware #103: SD Card Driver

jetkov, 2020-05-29:
- We should have a general SD card driver (do we need FS)

jetkov, 2020-05-29:
@a-gratton if you are going to be working on this as part of the GPS driver please assign yourself, otherwise this task is open for someone else, but you will need to sync with @a-gratton.

a-gratton, 2020-06-01:
I've assigned myself to this now

a-gratton, 2020-06-07:
**File system options:** 
We could use either [FATFileSystem](https://os.mbed.com/docs/mbed-os/v5.15/apis/fatfilesystem.html) or [LittleFileSystem](https://os.mbed.com/docs/mbed-os/v5.15/apis/littlefilesystem.html). 
**LittleFS** [uses less resources and is less likely to corrupt data](https://os.mbed.com/blog/entry/littlefs-high-integrity-embedded-fs/) on the card during an unexpected shutdown. However, it is [very difficult to work with outside of the mbed environment.](https://github.com/ARMmbed/littlefs/issues/38)
**FAT** is heaver in terms of resource use and is not as resilient as LittleFS. However, it is a well known standard and easy to work with on all operating systems.

Considering one of the use cases for the SD driver will be pre-loading satellite data for the GPS (and that our micro controllers are overspecialized for our needs), I think that FAT is the better choice. However it is pretty easy to switch between these APIs if we decide to go with LittleFS.

**Other APIs**
The [SDBlockDevice API](https://os.mbed.com/docs/mbed-os/v5.15/apis/sdblockdevice.html) is used by the filesystem API to access the SD card, but we don't really need to do anything with it besides pass it to the filesystem API.

wmmc88, 2020-06-07:
resiliency is probably better, especially because we always randomly shut off power,  and the user space program they linked in that thread looks straight forward to use.

a-gratton, 2020-06-08:
Also for interacting with the files we could go for C file IO (stdio) or C++ (fstream). C++ is easier to work with, but there seem to be [major](https://stackoverflow.com/questions/4340396/does-the-c-standard-mandate-poor-performance-for-iostreams-or-am-i-just-deali) [performance](https://stackoverflow.com/questions/37894262/c-iostream-vs-c-stdio-performance-overhead) [issues](https://stackoverflow.com/questions/605839/c-and-c-file-i-o) with it (although there is some [debate](https://stackoverflow.com/questions/2872543/printf-vs-cout-in-c) about that).

The other option would be to not actually write a file handling wrapper and just leave the filesystem exposed for people to do their own file IO, but then there isn't much of a point to the SD card library.

a-ebrahim, 2021-02-17:
What's the status on this?