uwrobotics/MarsRoverFirmware #104: LCD Driver (BMS)

jetkov, 2020-05-29:
LCD driver for SPI NHD-C12832A1Z-FSW-FBW-3V3 LCD on BMS board

sahil-kale, 2020-07-09:
Guess I'll try to tackle this one now, will probably end up flopping though.

jetkov, 2020-07-09:
No worries, please coordinate with @FK3wong as he is also working on BMS

FK3wong, 2020-07-09:
@jetkov Could you outline the goals for this LCD driver? For example are there specific things we want to display?

FK3wong, 2020-07-10:
> Guess I'll try to tackle this one now, will probably end up flopping though.

FK3wong, 2020-07-10:
@sahil-kale I think some first steps for you would be to get the data sheet and technical manual for this LCD board and maybe links to some examples and put them as part of this issue. That way we can easily start trying to understand the chip. If you aren't familiar with SPI I would recommend watching some videos or reading up on it as that will be how you will interface with the chip. Then you might want to create a header file which lists prototypes of useful functions and variables for using this device and put it up for review. Make sure to ask questions if you are stuck.

FK3wong, 2020-07-11:
I had a conversation with Ayesha about the BMS chip and the LCD screen and here is some more information about it:

Goal: Display real-time data about the batteries.
Information to be displayed may include critical data (E.g High current draw, Low battery levels) and the current status of the batteries. 
A nice to have could be the level of the battery in a visual similar to a battery bar on a smart phone home tab.

The LCD screen will likely be connected to the STM with the STM communicating with the BMS chip and the LCD separately. The STM get data from the BMS, process that data, then send the data to the LCD to display.

sahil-kale, 2020-07-12:
That makes sense, I'll drop some documentation and go from there.

(Got caught up in work a little, ill def contribute more though)

sahil-kale, 2020-07-23:
newhavendisplay.com/specs/NHD-C12832A1Z-FSW-FBW-3V3.pdf here's the datasheet

cindyli-13, 2021-06-04:
@FK3wong @sahil-kale Is this still an open issue? If not could you close it?

FelixKHWong, 2021-06-18:
Closed due to new BMS system project

closed 2021-06-18 by FelixKHWong