uwrobotics/MarsRoverFirmware #121: Add non-blocking (Async/DMA) SPI encoder decoding driver

jetkov, 2020-07-22:


younesr1, 2020-07-24:
I will get started on this after i merge my pid api pr.

younesr1, 2020-07-24:
could u explain this task a little more pls @jetkov

jetkov, 2020-07-27:
Right now just testing reading from the SPI encoder I have it synchronous so it is blocking, it needs to be made into a proper driver that is asynchronous and it may be benificial to use direct memory access (DMA) to have the SPI peripheral write the encoder value to a block of memory without CPU intervention.

We can work on this one together as the new Netzer encoders being shipped to me for testing/integration.

younesr1, 2020-07-28:
sounds good. I will get on it once the pidapi gets merged in which hopefully should happen soon.

younesr1, 2020-08-06:
I'll forfeit this in case any of the new members wanna take it on

jetkov, 2020-08-11:
This might not be the best first issue unless they have prior FW experience

closed 2021-02-23 by cindyli-13