uwrobotics/MarsRoverFirmware #126: Design Centralized Logging System

younesr1, 2020-08-06:
This is important for when we integrate all the various sub-systems. Without a central logging system, it would be inefficient and time consuming to jump between boards adding printf statements etc. Instead, we could have all sub-systems relay data about their current states to the jetson. Then we could connect to the jetson and filter these msgs. Alot has yet to be decided so there's a lot of room for  originality, but here are some rough requirements:

It should take advantage of mbed-trace. Ultimately, you will likely be adding a logger class that sits right above mbed-trace.
It should easily be compiled out when not testing (macro) as to not use resources during missions
On the jetson, we should be able to filter by log level and sub-system

younesr1, 2020-08-06:
i'm still deciding what protocol is most appropriate for sending the msgs. I do not think we should send them over CAN, because the integration testing we do tests CAN. 

I would recommend UDP/TCP. Ethernet cables are trivial to setup and take down. Mbed has libs for it. Downside is EE would need to add it to upcoming board revisions.

wmmc88, 2020-08-06:
What about just using the uart? It's already the default for print statements

wmmc88, 2020-08-06:
Although, we would probs need EE to make a board with a bunch of UARTs. That board could probably do things like label the msg with what board it's from, timestamp it, etc.

wmmc88, 2020-08-06:
For filtering the msgs, it might be nice to run a ros node that republishes the logs in a ros format and then just use the rqt_log gui

younesr1, 2020-08-06:
> Although, we would probs need EE to make a board with a bunch of UARTs. That board could probably do things like label the msg with what board it's from, timestamp it, etc.

Is your suggestion to connect all fw sub-systems to this "logger board" via uart and then connect the logger board to the jetson via uart? Why not skip the part where we connect it to the jetson? We can just connect to the logger board and see all the msgs on it. If we do this, we can have some script that takes in the desired filtering and displays appropriate data.

But I guess the nice thing abt routing the data to the jetson is you can run a ros  node as you mentioned. Also, you wouldn't have to be tethered to the rover to read the log msgs, you can just ssh in.

younesr1, 2020-08-06:
uart is a good option too. it'd be twice the wiring that we would've needed with ethernet. Ethernet approach also means log msgs end up on Jetson. This might not be the case with the uart approach based on the above comment.

wmmc88, 2020-08-06:
@younesr1 connect to jetson so we can see logs from any computer on the network. Live and also can  bag em to diagnose issues easier

wmmc88, 2020-08-06:
@younesr1 not enough ports on the edge router for every board. Also uart is only one twisted pair, wheres ethernet is 4 twisted pairs.

younesr1, 2020-08-06:
@wmmc88 To solve the issue of not enough ports, i thought we could buy a cheap ethernet switch (~15$) that just repeats all traffic to everyone connected. It wouldnt clutter the system at mission time since we'd remove the entire thing.

Regardless, I think we are better off with uart anyway.

younesr1, 2020-08-11:
@wmmc88 what about using USART? Benefit is it can transmit data much faster. Downside is that we would likely need ee to expose a usart clock pin.
I dont think its worth it. what do you think?

wmmc88, 2020-08-11:
With usart you can also just encode the clock on the data line. Need to check that mbed supports it somewhere.

wmmc88, 2020-08-11:
Also with these changes, would the normal uart still work? How many uart perifs are there?

younesr1, 2020-08-11:
there are 2 uarts and 4 usarts. The usarts can be used as uarts but uarts cant be used as usart. For this reason, i say we use uart

jetkov, 2020-08-11:
imo CAN or UART. remember we have 2 CAN interfaces on each board. Are you thinking about having this run all the time or just for debug?

For minimal error logging / health monitoring there are some CAN IDs allocated for this already but has not been implemented (intent for full-time use). @wmmc88 and I also talked in the past about heartbeats to monitor the CAN bus state. 

Can give more info after finals or @wmmc88 can explain.

wmmc88, 2020-08-11:
@jetkov i think the health monitoring and can heartbeat msgs are full time things that should run on the can bus. That said, I think the purpose of this issue is different. 

I think the goal of this system is a way to basically aggregate prints from fw and:
1. send to ros so can be viewed in realtime
2. used to debug issues after the fact

The intent would be to have this running all the time, but its functionality is not critical. I don't think having it on the can bus is worth it because the nature of these log messages is that theyll be long strings and we don't want to have those slowing down can comms.

wmmc88, 2020-08-12:
https://stm32f4-discovery.net/2017/07/stm32-tutorial-efficiently-receive-uart-data-using-dma/

younesr1, 2020-08-13:
I shared this plan with EE. I believe they are getting started on it soon. I think someone can take on the library design part of this task . They should bring in mbed-trace and create the logger class and not write the fw for the logger board yet. Once we decided more about how the logger board will work, we'll make an APP and TARGET for that.
We still need to decide on a couple things:
1. What are the log levels?
2. Who can publish log messages? Can libraries or should only main.cpp files
3. Who timestamps? Do we timestamp at msg creation, at reception on logger board, or at reception on jetson.
4. What will filtering on Jetson look like?
Also, any one you guys know examples of well designed loggers i can look at?

younesr1, 2020-08-13:
My answers to the above questions:
1. ERROR(asking to set neopixels to unknown state) WARNING(not sure of example) STATEINFO(setting neopixel to green) CANINFO(received/sent this can msg). If at level n, we also see levels <n. for example, if at Warning level, we would see warning and error
2. Only main.cpp files
3. jetson as to not send unnecessary data.
4. We can list out the boards we wanna see msgs from and also specify the log level

wmmc88, 2020-08-13:
> Available levels:
> 
> debug
> info
> warning
> error
> cmdline (special behavior, should not be used)
~from [mbed-trace](https://github.com/ARMmbed/mbed-trace)

maybe we should match these levels

wmmc88, 2020-08-13:
not sure if it makes sense to change the led colours everytime a msg is sent lol

qinyang-bao, 2020-08-13:
Might not be too related, but part of this project sounds like what we did with the custom robot controller board --- there is a master board connected to pc and all other boards, and msgs are sent via UART....

I think both could share some library for aggregating & re-transmitting UART msgs

wmmc88, 2020-08-13:
>3. Who timestamps? Do we timestamp at msg creation, at reception on logger board, or at reception on jetson.

I think the logger board should timestamp the msgs. this way its closer to the actual timestamp, although not as close as on the individual boards themselves. Having the logger board timestamp however will all for a common reference point for all the timestamps because each board will have a different zero. This is because it will be a timestamp relative to board bootup, no realtime. 

If we wanted to, we could add a rtc to the logger board for absolute timestamps, but i don't think we need that. I've built a similar system like this before and we could use a similar approach of converting the relative timestamp from the logger board to a absolute timestamp once the jetson receives it by looking at the offests

wmmc88, 2020-08-13:
> What will filtering on Jetson look like?
I think we can convert logs to ros console logs, tagged by board and mapping mbed trace levels to ros debug levels. ros debug levels can easily be filtered (including thru rosconsole rqt gui)

wmmc88, 2020-08-13:
> I shared this plan with EE. I believe they are getting started on it soon. I think someone can take on the library design part of this task . They should bring in mbed-trace and create the logger [...]

I think that this project should have a high level plan drawn out asap so that the important reqs are flushed out (before ee starts building something that may be missing critical features)

younesr1, 2020-08-13:
I like the idea of using the same log levels as mbed-trace. If the logger board timestamps, i'd say stick an rtc on it. i believe it'd be low effort and low cost. 

I will put together a high level plan drawing before tomorrow's technical meeting hopefully.

wmmc88, 2020-08-14:
The (small) problem with having an rtc with the logger board is that now there are 2 sources of truth for the absolute timestamps, the jetson clock and the logger clock. How do we deal with syncing them up? Which time is used inside the rosconsole log? What happens when the two dont match? 

Just things to think about

younesr1, 2020-08-18:
Our master clock is the rtc on the logger. i say that we dont care about the time shown on the jetson. We do not sync them up. 
Here's a requirement doc i've started working on. it is not yet complete

https://uofwaterloo-my.sharepoint.com/:w:/g/personal/yreda_uwaterloo_ca/ER5gQUd_c5tBtht4vzOmRVwB5HSMQFVFnM8q-1ZHKrfJpw?e=LRTUzG

wmmc88, 2020-08-18:
I think the problem with ignoring the jetson clock is that if we bag data, stuff isnt gonna be in sync. I think there needs to be a handshake somewhere to account for the difference.

younesr1, 2020-08-26:
This task is closed. The final design is the following:
Use the secondary CAN bus to transmit log messages. The log system will be live during testing and at mission time.
Log msgs should carry information about board of origin and log level and call stack.
If the primary CAN bus fails and we start using the seconday CAN bus, we abandon the logger
For longer msgs, consider stitching them together.
Consider using rolling counter to ensure log msgs from each board are kept in sync.

closed 2020-08-26 by younesr1