uwrobotics/MarsRoverFirmware #141: Implement Logger

younesr1, 2020-08-26:
We would like to have a centralized logging system. The idea is that all sub-systems use this logger to report warnings errors etc which are received by the Jetson and bagged for looking back at a later point. Below are some specifics, but whoever picks this up can reach out to me for a call.

Use the secondary CAN bus to transmit log messages. The log system will be live during testing and at mission time.
Log msgs should carry information about board of origin and log level.
If the primary CAN bus fails and we start using the seconday CAN bus, we abandon the logger
For longer msgs, consider stitching them together.
Consider using rolling counter to ensure log msgs from each board are kept in sync.

wmmc88, 2021-01-03:
Should maybe consider moving to a more C++-like iostream-type prints instead of printfs. https://github.com/uwrobotics/MarsRover2020-firmware/pull/231#discussion_r550938952

cindyli-13, 2021-06-04:
Closing this as logger is already implemented

closed 2021-06-04 by cindyli-13