uwrobotics/MarsRoverFirmware #142: Building uses too many resources

younesr1, 2020-08-26:
Building causes machines to become very slow. Perhaps use $NPROC/2 for number of threads used.

wmmc88, 2020-08-26:
building is supposed to use as many threads as possible(up to a max of cores in ur computer x 2) so that it finishes as fast as possible. basically every build system is like this. ex. catkin in ros also uses nproc as default amount of threads used.

wmmc88, 2020-08-26:
The build process in our fw appears to  use more cores on purpose. Not all build sequences can be fully parallizeable and effort needs to be put into the design of targets so that it can have optimally parallel builds. the goal is always to have all available cpus at 100% so that no resources are wasted. The fact that fw consumes more than the ros build simply means that the fw build is more parallel and optimized more.

wmmc88, 2020-08-26:
if people really really want to reduce cpu usage, u could add a flag to the wrapper make to manually specify max number of jobs. 

idk why anyone would ever want to slow down their builds tho... the most complex target (arm, arm) takes ~5 seconds to build from clean on my computer and 5 seconds wait is a hell of a lot better than a 30 second wait.

younesr1, 2020-08-26:
i agree that performance is very fast. the issue is that it takes up too many resources. it freezes my machine until building is done.
i suggest adding a flag called --slow that uses have the cores.

wmmc88, 2020-08-26:
how long is frozen for?

younesr1, 2020-08-26:
like a minute. sometimes it just hangs for longer and i just end up giving up and rebooting my computer. ill bring it up in the meeting to see how many ppl are in favor of --slow. if more ppl are in favor of --slow, ill make that the default and --fast the flag

wmmc88, 2020-08-26:
did u have minute long freezes with make? cuz that was also using -nproc

younesr1, 2020-08-27:
No, definitely not. it was only with cmake.

wmmc88, 2020-08-27:
Can you capture the actual resources used during build? I suspect it's not the cpu usage that causes the issue. It's may be hitting the max physical ram and then your computer starts constantly switching to data to swap

PR #161 (2020-09-24): Makefile Enhancements
closes #142 
closes #135
commit 9885974f (2020-09-24): Makefile Enhancements (#161)

* closes #142 add option to reduce build resources

* closes #135: add "make all" cmd to build all supported app/target configs
closed 2020-09-24 by wmmc88