uwrobotics/MarsRoverFirmware #144: Add anti-windup to PID class

younesr1, 2020-09-04:
Anti windup could potentially be crucial at mission time. If the arm tries interacting with something that becomes stuck, we could have undesried consequences if we do not guard against integral wind-up.

Essentialy, integral windup is when your integral path balloons because your error is not decreasing for some reason. It balloons beyond your max output. So now you have a large Integral output even after you are past your set point so you overshoot.

Here's a video explaining it (2nd half): https://www.youtube.com/watch?v=NVLXCwc8HzM&t=3s&ab_channel=MATLAB
Here's a PID class implementing it: https://github.com/mick001/PID-Controller/blob/master/C%2B%2B_code/pid.cpp

FK3wong, 2020-09-15:
@younesr1 Can I get a date as a deadline for this?

FK3wong, 2020-09-22:
There exists an implementation of anti-windup inside the current PID class. On lines 414-415 of function PID::compute(), there exists a check to determine if the PID output is outside the expected bounds. In this event the Integral's accumulated error will be frozen and will no longer accumulate error until the PID output drops into the acceptable range.

closed 2020-09-22 by FK3wong