uwrobotics/MarsRoverFirmware #149: Update ActuatorController to optionally use cascaded PID

younesr1, 2020-09-15:
This video explains cascaded PID control: youtube.com/watch?v=tbgV6caAVcs&t=187s&ab_channel=MATLAB

This task requires modifying the ActuatorController class to optionally take in a second PID to implement cascaded PID. I'd recommend becoming familiar with the ActuatorController class and its children as well as our PID class before starting to write code. For this task, planning is really important.

Backstory:
We want the arm to use cascaded PID with the inner loop being torque (equivalent to current : https://www.youtube.com/watch?v=WncK1qbEgYE&ab_channel=TimCallinan) control. The torque control allows us to do fancy controls like implementing a feed forward term. The outer loop is velocity PID.

cindyli-13, 2020-09-20:
I have some general questions related to this task (just gonna dump them here):

- Will the cascaded control mainly be used for shoulder or all arm actuators?
- (Probably a dumb question) Do encoders give > 360 deg readings?
- How are PID parameters currently tuned? How would cascaded PID control be tuned (do we know yet)?
- Will we ever want to also use inner loop = torque, outer loop = position (for cascaded PID control)?
- How does torque feed forward control work?
- Do we want the ability to switch between a single PID loop and cascaded PID during operation or are we deciding beforehand which control configuration to use for a specific actuator?

wmmc88, 2020-09-20:
- all
- they roll over at 360 afaik
- rn guess and check LOL. we also have an autotune library we could try to use. imo we should just send it some step signal etc and graph the response and tune it that way(so guess and check with some more precise checking basically). we would need to tune the inner pid first, then tune the outer one. This might require the firmware being able to accept current commands.
- potentially yes. i would build it into the design maybe... maybe have its so that a parameter can toggle whether or not there is an inner torque pid? in terms of levels of abstraction, i think its safe to say that we wont ever have more than just the 1 cascade.
- the velocity pid will output a desired current command for the given joint. the ff term at that instant would be computed from the given joint's velocity and position (and intertia). this term is equivalent to torque needed to compensate to effect of gravity. The term should be multiplied by a motor-specific constant(to convert to current @MECH/ELEC WE NEED THESE CONSTANTS) and then added/subtracted to the current command. 
- on the fly switch could be a feature.

wmmc88, 2020-09-20:
@younesr1 it might be good to have that actuator firmware high level rearchitecture/retrospective to see how all these new features should be implemented.

younesr1, 2020-09-24:
Yes definetly, I havent gotten to doing that yet.

cindyli-13, 2021-06-04:
Closing this as we're using ODrive PID control instead

closed 2021-06-04 by cindyli-13