uwrobotics/MarsRoverFirmware #154: Turn all doubles to floats

younesr1, 2020-09-17:
We use the M4 processor on our boards. With such processors, dividing floats is rather difficult. It actually takes alot of effort. We are fortunate that our processor has a Floating Point Unit (fpu), which accelerates such operations.

However, this acceleration of floating point arithmetic is only limited to floats. It does not apply to doubles. This is something we only learned recently. For this reason, the repo should use floats over doubles when possible because the FPU will make the arithmetic of a float run much faster than the arithmetic of a double.

Read this link to understand more: http://support.raisonance.com/content/how-should-i-use-floating-point-unit-fpu-cortex-m4

Ultimately, this task boils down to changing doubles to float when appropriate. But I highly recommend you take the time to really understand the reason for this task. Read into a FPU and how it is made up etc

younesr1, 2020-09-17:
@niiquaye

niiquaye, 2020-09-17:
Alright. Will do. Lastly, you said we only forked the repository in the training but we are actually working on issues we make branches. Is this correct ?

younesr1, 2020-09-17:
yes, there is a detailed naming convention to name ur branch in the readme

niiquaye, 2020-09-17:
alright thanks

commit d795385f (2020-09-24): Closes #154: change all doubles to floats (#156)

* First Commit

* Clang Format Error Fix - Second Commit

* Clang Error Fix - Third Commit

* Fourth Commit - Changed More Doubles to Floats

* Fifth Commit

* Sixth Commit - Melvin's Requested Changing of Comments
closed 2020-09-24 by niiquaye