uwrobotics/MarsRoverFirmware #164: Create current sensor class

younesr1, 2020-09-27:
We need to add a current sensor class to our lib.

younesr1, 2020-09-27:
@a-gratton assign urself to this if you'll take this on pls

younesr1, 2020-09-27:
https://www.melexis.com/-/media/files/documents/datasheets/mlx91205-datasheet-melexis.pdf

FK3wong, 2020-09-27:
@younesr1 is this related to the INA-226 in any way?

wmmc88, 2020-09-28:
@FK3wong sort of lol. We're replacing the INAs with this lol

FK3wong, 2020-09-28:
Are we still going to use the INA-226 for anything then

younesr1, 2020-09-28:
this is for the current sensors on the arm which are not ina-226. the data sheet is linked above. bms might still be restarted at some point but we dont know yet

a-gratton, 2020-10-30:
School ended up being really busy for a while, I am assigning this to myself now.

wmmc88, 2021-02-26:
Just a fyi, we'll need to make sure the driver reads rms current since it's a pwm wave