uwrobotics/MarsRoverFirmware #305: Add hardware filtering

younesr1, 2021-03-01:
This is post SAR critical! It will cause apps to crash if not implemented

wmmc88, 2021-03-01:
I think should also not crash even if it receives smth weird. just log it

younesr1, 2021-03-11:
@cindyli-13 can u assign this to urself? i think ur moving this stuff from sar branch to master?

cindyli-13, 2021-03-11:
HW filtering is already added to master, but should the lookup table itself handle cases where the key doesn't exist?

wmmc88, 2021-03-11:
It should be a logger print, not a run time fault. Espc since I dont think he filtering will 
be able to filter out all the unhandled ids

younesr1, 2021-03-11:
does the he filtering filter out all the stuff thats not relevant for the 4 main apps? if so, we shuold switch MBED_ERROR for MBED_WARN if not, then just return unsupported

wmmc88, 2021-03-12:
> does the the filtering filter out all the stuff thats not relevant for the 4 main apps? if so, we shuold switch MBED_ERROR to MBED_WARN if not, then just remove the MBED_ERROR. no need to print.         MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_APPLICATION, MBED_ERROR_CODE_INVALID_ARGUMENT),

Can we discuss this in tonight's meeting?

closed 2021-03-25 by cindyli-13