uwrobotics/MarsRoverFirmware #312: Add units library to repo

younesr1, 2021-03-05:
Investigate using a units library like this
https://github.com/nholthaus/units

it would simplify function names like readAngleDegPerSec() and constants like maxRPM, and will generally be safer

@wmmc88 @cindyli-13 thoughts? this could be an easy task for new members

wmmc88, 2021-03-05:
I'd prefer to just standardize units instead. Like ros has a REP that standardizes everything.

cindyli-13, 2021-03-05:
I'm fine with either units library or standardizing units, as long as we're consistent throughout the whole repo

younesr1, 2021-03-05:
@upadhyaydhruv for starters, create a units library and write a test-units app the shows how to use these units: Amperes, Degrees, Degrees per sec, Newtons