uwrobotics/MarsRoverFirmware #404: Update cmake to use mbed_create_distro and latest mbed release

wmmc88, 2022-01-07:
This repo currently uses a fork of mbed feauture branch `feature-cmake-object-libraries` that was patched by me to pull in a fix for STM32 targets(which has since been officially released upstream), and then patched by @cindyli-13 to provide several can fixes(I don't think these fixes were merged upstream but @cindyli-13 can confirm). 

According to discussion in https://github.com/ARMmbed/mbed-os/issues/13981, it seems like upstream mbed has gone a different direction and it looks like the `feature-cmake-object-libraries` is not going to be merged into mainline. Instead, it looks like mainline will be using the `mbed_create_distro` feature that merged in https://github.com/ARMmbed/mbed-os/pull/15126 instead.

We should probably pull in the latest official upstream release (patched with our can fixes), and port our repo to use `mbed_create_distro`. As it stands, our fw repo is using a version of mbed thats missing ~a years worth of bugfixes, features, etc

0xc0170, 2022-01-10:
> According to discussion in ARMmbed/mbed-os#13981, it seems like upstream mbed has gone a different direction and it looks like the feature-cmake-object-libraries is not going to be merged into mainline. Instead, it looks like mainline will be using the mbed_create_distro feature that merged in ARMmbed/mbed-os#15126 instead.

It would be the best to use `mbed_create_distro`. The feature branch is already way behind and it won't be integrated anytime soon. Did the feature branch work for you? i wonder how much effort it is to update to the latest release and use this create distro function instead.

cindyli-13, 2022-01-14:
> 
The CAN patch was merged upstream actually in https://github.com/ARMmbed/mbed-os/pull/14612

wmmc88, 2022-01-15:
> > This repo currently uses a fork of mbed feauture branch `feature-cmake-object-libraries` that was patched by me to pull in a fix for STM32 targets(which has since been officially released upstream), and then patched by @cindyli-13 to provide several can fixes(I don't think these fixes were merged upstream but @cindyli-13 can confirm).
> > According to discussion in [ARMmbed/mbed-os#13981](https://github.com/ARMmbed/mbed-os/issues/13981), it seems like upstream mbed has gone a different direction and it looks like the `feature-cmake-object-libraries` is not going to be merged into mainline. Instead, it looks like mainline will be using the `mbed_create_distro` feature that merged in [ARMmbed/mbed-os#15126](https://github.com/ARMmbed/mbed-os/pull/15126) instead.
> > We should probably pull in the latest official upstream release (patched with our can fixes), and port our repo to use `mbed_create_distro`. As it stands, our fw repo is using a version of mbed thats missing ~a years worth of bugfixes, features, etc
> 
> The CAN patch was merged upstream actually in [ARMmbed/mbed-os#14612](https://github.com/ARMmbed/mbed-os/pull/14612)

So we should be able to go back to using mainline mbed, right? no other patches we made that havent been upstreamed already?

cindyli-13, 2022-01-16:
> 

As far as I'm aware yes. Unless there were patches other than the CAN patch that aren't in mainline.